borisbu/OctoRelay #270: BIGTREETECH BTT  Pi V1.2  The plugin is installed but it is not in the plugin manager.

Aleks-MO, 2024-05-14:
On BIG TREE TECH BTT Pi V1.2 from debian 11 taken from github - CB1_Debian11_minimal_kernel5.16_20240319.img.xz     Octoprint 1.10 
The plugin is installed but it is not in the plugin manager. In protected mode, I see this plugin. What to do?

RobinTail, 2024-05-14:
I'd ask for support on OctoPrint forum, @Aleks-MO

RobinTail, 2024-05-14:
Perhaps it's a file permission issue. Or there may be other reasons why OctoPrint's plugin manager does not list an installed plugin, @Aleks-MO

Aleks-MO, 2024-05-14:
@RobinTail  All GPIO-related plugins behave like this, there are no problems with the rest. ((((

RobinTail, 2024-05-14:
> All GPIO-related plugins

Then I'd assume that the issue is related to the `RPi.GPIO` module in your environment.

Could you please enable DEBUG level for OctoRelay in OctoPrint logging settings, restart it, and try finding the entries related to OctoRelay.

Perhaps it's the issue similar to #255 , @Aleks-MO

RobinTail, 2024-05-14:
Just in case you're using several GPIO related plugins, if other plugins have such a setting as pins enumeration, please ensure that they all are configured to use `BCM` one.
But logs could say more if there are any actual errors in your runtime, @Aleks-MO

Aleks-MO, 2024-05-14:
Here is the old octoprint log

2024-04-22 22:25:05,916 - octoprint.plugins.pluginmanager - INFO - Installing plugin from https://github.com/borisbu/OctoRelay/releases/latest/download/release.zip
2024-04-22 22:25:07,334 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin (2023.10.10) did not pass check, not loading.
2024-04-22 22:25:29,441 - octoprint.plugin.core - ERROR - Error loading plugin octorelay
Traceback (most recent call last):
  File "/home/biqu/OctoPrint/venv/lib/python3.9/site-packages/octoprint/plugin/core.py", line 1294, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/biqu/OctoPrint/venv/lib/python3.9/site-packages/octoprint/plugin/core.py", line 52, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/biqu/OctoPrint/venv/lib/python3.9/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/home/biqu/OctoPrint/venv/lib/python3.9/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/biqu/OctoPrint/venv/lib/python3.9/site-packages/octoprint_octorelay/__init__.py", line 20, in <module>
    from .driver import Relay
  File "/home/biqu/OctoPrint/venv/lib/python3.9/site-packages/octoprint_octorelay/driver.py", line 3, in <module>
    from RPi import GPIO
  File "/home/biqu/OctoPrint/venv/lib/python3.9/site-packages/RPi/GPIO/__init__.py", line 23, in <module>
    from RPi._GPIO import *
RuntimeError: This module can only be run on a Raspberry Pi!

RobinTail, 2024-05-15:
Found a user manual here. And there is a repo that probably can contain even more information 

https://github.com/bigtreetech/BTT-Pi/blob/master/BIGTREETECH%20Pi%20V1.2%20User%20Manual.pdf

RobinTail, 2024-05-15:
In the end, @Aleks-MO , I recommend to raise an issue in https://github.com/bigtreetech/BTT-Pi in order to ask what should be the replacement of "RPi.GPIO" python module in your case.

Aleks-MO, 2024-05-15:
@RobinTail  Yes, this board and description, there is also a Debian 11 build - https://github.com/bigtreetech/CB1/releases/tag/V2 .3.4 which is menial. But it's not clear how it works with GPIO.

RobinTail, 2024-05-15:
Related Reddit (no solution): https://www.reddit.com/r/BIGTREETECH/comments/182b8op/btt_pi_python_gpio/

Or maybe you can switch to Klipper which is your hardware seems to be optimized for (instead of OctoPrint), @Aleks-MO ?

RobinTail, 2024-05-15:
Found this (another board from same vendor):
https://github.com/bigtreetech/CB1/issues/66

Aleks-MO, 2024-05-15:
I've seen it, it's not the first question for BTT. It doesn't work either -https://wiki.radxa.com/Gpiod

Aleks-MO, 2024-05-15:
@RobinTail Thanks!, I didn't see the last link. I'll try to write here, it helped or not. )

RobinTail, 2024-05-15:
Yeah, in order to address it properly, we make another driver for your board

https://github.com/borisbu/OctoRelay/blob/f323c978d52986060ffbe5db93f62cf5d3f2e991/octoprint_octorelay/driver.py#L3

Using this module instead:

https://pypi.org/project/gpiod/

It seems to have a completely different interface and requires to define the "GPIO chip character device".

Aleks-MO, 2024-05-16:
The "gpiochip0" is used with the legs "Gpio 0 to 287". Pin 211 is installed on the fan control board. Turning on and off the light pins 203 and 201. I don't understand how to prescribe it. (((

biqu@BTT-CB1:~$ sudo cat /sys/kernel/debug/gpio
gpiochip0: GPIOs 0-287, parent: platform/300b000.pinctrl, 300b000.pinctrl:
 gpio-166 (                    |:wifi_power         ) out hi
 gpio-207 (                    |:wifi_wake          ) out hi
 gpio-210 (                    |reset               ) out hi ACTIVE LOW
 gpio-229 (                    |sysfs               ) out hi

gpiochip1: GPIOs 352-383, parent: platform/7022000.pinctrl, 7022000.pinctrl:
biqu@BTT-CB1:~$ gpiodetect
gpiochip0 [300b000.pinctrl] (288 lines)
gpiochip1 [7022000.pinctrl] (32 lines)
iqu@BTT-CB1:~$ gpioinfo
gpiochip0 - 288 lines:
        line   0:      unnamed       kernel   input  active-high [used]
        line   1:      unnamed       kernel   input  active-high [used]
        line   2:      unnamed       kernel   input  active-high [used]
        line   3:      unnamed       kernel   input  active-high [used]
        line   4:      unnamed       kernel   input  active-high [used]
        line   5:      unnamed       kernel   input  active-high [used]
        line   6:      unnamed       kernel   input  active-high [used]

Aleks-MO, 2024-05-18:
@RobinTail  Here is the link given by the BTT manufacturer https://blog.csdn.net/hsl416604093/article/details/122634479 ?
I tried to turn on and off the fan on the board, it works. Can you fix the plug-in for the BTT Pi v2 board?  

biqu@BTT-CB1:~$ cd /sys/class/gpio/
biqu@BTT-CB1:/sys/class/gpio$ ls
export  gpio229  gpiochip0  gpiochip352  unexport
biqu@BTT-CB1:/sys/class/gpio$ sudo su
root@BTT-CB1:/sys/class/gpio# sudo echo 211 > export
root@BTT-CB1:/sys/class/gpio# ls
export  gpio211  gpio229  gpiochip0  gpiochip352  unexport
root@BTT-CB1:/sys/class/gpio# sudo echo out > direction
bash: direction: Permission denied
root@BTT-CB1:/sys/class/gpio# cd /sys/class/gpio/gpio211
root@BTT-CB1:/sys/class/gpio/gpio211# sudo echo out > direction
root@BTT-CB1:/sys/class/gpio/gpio211# sudo echo 1 > value
root@BTT-CB1:/sys/class/gpio/gpio211# ls
active_low  device  direction  edge  power  subsystem  uevent  value
root@BTT-CB1:/sys/class/gpio/gpio211# sudo echo 0 > value
root@BTT-CB1:/sys/class/gpio/gpio211# cd /sys/class/gpio/
root@BTT-CB1:/sys/class/gpio# sudo echo 211 > unexport
root@BTT-CB1:/sys/class/gpio#

RobinTail, 2024-05-18:
Does it work without `sudo`, @Aleks-MO ?

RobinTail, 2024-05-18:
What they gave you is not a python library, but an approach on how to operate GPIO using file system via CLI commands.

I could possibly make a special driver for you that executes some system commands, but I anticipate that it will work significantly slower. If the commands can be executed without `sudo`, then maybe that driver could read/write to those "device files" directly.

But all that, I must say it, appears very weird to me so far.
The [gpiod approach](https://github.com/borisbu/OctoRelay/issues/270#issuecomment-2112873353) looks more promising. But they both require an effort and investigation, @Aleks-MO

RobinTail, 2024-05-18:
@Aleks-MO , I checked that `gpiod` library works in general on my Raspberry Pi so I could probably make a driver based on it. Could you please try the following:

1. `pip3 install gpiod`
2. And tell me which version would be installed in your case
3. Make a file `test_gpiod.py` of the following content:

```
import gpiod

exists = gpiod.is_gpiochip_device("/dev/gpiochip0")

print(f"Exists {exists}")

chip = gpiod.Chip("/dev/gpiochip0")
info = chip.get_info()

print(f"{info.name} [{info.label}] ({info.num_lines} lines)")
```

4. And then run in `python3 test_gpiod.py` and tell me what it says, should be similar to this:

```python
Exists True
gpiochip0 [pinctrl-bcm2835] (54 lines)
```

RobinTail, 2024-05-18:
There is also this package, could also work
https://python-periphery.readthedocs.io/en/latest/gpio.html

Used by PSU Control: https://github.com/kantlivelong/OctoPrint-PSUControl/blob/master/octoprint_psucontrol/__init__.py#L23

Aleks-MO, 2024-05-19:
I did this test, the result is negative.
Last login: Fri May 17 12:54:56 2024 
biqu@BTT-CB1:~$ pip3 install gpiod
Collecting gpiod
  Downloading gpiod-2.1.3.tar.gz (53 kB)
     |████████████████████████████████| 53 kB 104 kB/s
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Building wheels for collected packages: gpiod
  Building wheel for gpiod (PEP 517) ... done
  Created wheel for gpiod: filename=gpiod-2.1.3-cp39-cp39-linux_aarch64.whl size=91494 sha256=bd8cbfccab914b76906bdceb3c4ef28014661b3bbe0b1ea1e3e0c30a6ed4e4bf
  Stored in directory: /home/biqu/.cache/pip/wheels/5d/18/00/aff28d894b61a5b1a0a230fc8099f8eb7ad6343d344e159b66
Successfully built gpiod
Installing collected packages: gpiod
Successfully installed gpiod-2.1.3
biqu@BTT-CB1:~$ test_gpiod.py
test_gpiod.py: command not found
biqu@BTT-CB1:~$ sudo su
root@BTT-CB1:/home/biqu# test_gpiod.py
test_gpiod.py: command not found
root@BTT-CB1:/home/biqu# sudo python3 test_gpiod.py
Traceback (most recent call last):
  File "/home/biqu/test_gpiod.py", line 1, in <module>
    import gpiod
ModuleNotFoundError: No module named 'gpiod'
root@BTT-CB1:/home/biqu#

RobinTail, 2024-05-19:
Please run without `sudo`, @Aleks-MO 

```sh
python3 test_gpiod.py
```

RobinTail, 2024-05-19:
@Aleks-MO , you ran `pip3` as `biqu`, but you're running `python` as `root`.
I believe it causes the issue.

Aleks-MO, 2024-05-19:
Without SUDO, it looks like this

biqu@BTT-CB1:~$ test_gpiod.py
test_gpiod.py: command not found

Aleks-MO, 2024-05-19:
@RobinTail  Yes, I started it from the user BIQU.

Last login: Sun May 19 09:32:04 2024 
biqu@BTT-CB1:~$ python3 test_gpiod.py
Exists True
gpiochip0 [300b000.pinctrl] (288 lines)
biqu@BTT-CB1:~$

RobinTail, 2024-05-19:
> biqu@BTT-CB1:~$ test_gpiod.py

The word `python3` is missing.

> `gpiochip0 [300b000.pinctrl] (288 lines)`

great!
So this library works with your board, @Aleks-MO 
I'm making a driver based on this library in the following branch: #271 

Once I test it myself, I will let you know how you can join me.

Aleks-MO, 2024-05-19:
@RobinTail  Yes, there was my inattention... Apologize. )

Thank you very much, I would appreciate it.

RobinTail, 2024-05-20:
I made a version that demonstrates some sort of vitality, @Aleks-MO 
You can try installing it from this link:
```
https://github.com/borisbu/OctoRelay/releases/download/5.0.0a3/release.zip
```

However, it has some issues. In particular, it does not track in background the changes happened to the relay state correctly.

It seems that using the `gpiod` library can not be done in a drop-in way, but rather requires a complete refactoring because it has a different logic of the operation.

Nevertheless, you can check if at least it could do something with your board.

P.S. I have no idea what pin numbers should be used to address your relays correctly. If there are alternative enumerations — try different ways.

Aleks-MO, 2024-05-20:
@RobinTail The plugin has appeared in the plugin manager, the settings are available. There are no icons in the upper tray. After rebooting, the octoprint takes a long time to load or freezes. Does not work.
Maybe you're right, it won't work properly. After removing the plugin, the octoprint loads normally, there are no freezes. Maybe it's really time to switch to the Clipper. The verdict on CB1 and Piv 2 does not work.
[image]
[image]

Aleks-MO, 2024-05-20:
@RobinTail Thank you for your efforts.   It's a pity that it didn't work.

RobinTail, 2024-05-20:
Alright. Yes, it seems to require more effort than I currently could dedicate.
Thank you for collaboration, @Aleks-MO

closed 2024-05-20 by RobinTail

RobinTail, 2024-07-10:
@Aleks-MO , in case you're still using Octoprint, we made the Octorelay version 5 that uses a more versatile GPIO driver that supports Raspberry Pi 5 and may be compatible with your hardware as well.