borisbu/OctoRelay #284: Update to OctoRelay 5.0.0 - 'swig': No such file or directory

Orlando522, 2024-07-11:
When I tried to update OctoRelay to version 5.0.0 I got error: unable to execute 'swig': No such file or directory.
Whole log:

Updating, please wait.

+++++++++++++++++++++++++++++++
Now updating OctoRelay to 5.0.0
+++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install https://github.com/borisbu/OctoRelay/releases/download/5.0.0/release.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/borisbu/OctoRelay/releases/download/5.0.0/release.zip
Downloading https://github.com/borisbu/OctoRelay/releases/download/5.0.0/release.zip (51 kB)
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from OctoRelay==5.0.0) (1.10.2)
Requirement already satisfied: RPi.GPIO in /home/pi/oprint/lib/python3.7/site-packages (from OctoRelay==5.0.0) (0.7.1)
Collecting gpiozero
Downloading https://www.piwheels.org/simple/gpiozero/gpiozero-2.0-py3-none-any.whl (150 kB)
Requirement already satisfied: colorzero in /home/pi/oprint/lib/python3.7/site-packages (from gpiozero->OctoRelay==5.0.0) (2.0)
Requirement already satisfied: setuptools in /home/pi/oprint/lib/python3.7/site-packages (from colorzero->gpiozero->OctoRelay==5.0.0) (51.1.2)
Collecting lgpio
Downloading lgpio-0.2.2.0.tar.gz (90 kB)
Requirement already satisfied: tornado<6.3,>=6.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (6.2)
Requirement already satisfied: pytz in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (2023.3)
Requirement already satisfied: flask<2.3,>=2.2.3 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (2.2.5)
Requirement already satisfied: pydantic==1.10.16 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (1.10.16)
Requirement already satisfied: Flask-Assets<3,>=2.1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (2.1.0)
Requirement already satisfied: blinker<1.7.0,>=1.6.3 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (1.6.3)
Requirement already satisfied: wheel in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (0.36.2)
Requirement already satisfied: Babel<2.13,>=2.12.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (2.12.1)
Requirement already satisfied: netaddr<0.9,>=0.8 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (0.8.0)
Requirement already satisfied: netifaces<0.12,>=0.11.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (0.11.0)
Requirement already satisfied: pathvalidate<3,>=2.5.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (2.5.2)
Requirement already satisfied: OctoPrint-PiSupport>=2023.10.10 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (2023.10.10)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (2021.10.11)
Requirement already satisfied: Flask-Login<0.7,>=0.6.3 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (0.6.3)
Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (2024.3.27)
Requirement already satisfied: requests<3,>=2.31.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (2.31.0)
Requirement already satisfied: wrapt<1.16,>=1.15 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (1.15.0)
Requirement already satisfied: markdown<3.5,>=3.4.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (3.4.4)
Requirement already satisfied: sarge==0.1.7.post1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (0.1.7.post1)
Requirement already satisfied: websocket-client==1.6.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (1.6.1)
Requirement already satisfied: cachelib<0.11,>=0.10.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (0.10.2)
Requirement already satisfied: filetype<2,>=1.2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (1.2.0)
Requirement already satisfied: argon2-cffi>=23.1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (23.1.0)
Requirement already satisfied: pkginfo<2,>=1.9.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (1.9.6)
Requirement already satisfied: zeroconf~=0.127 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (0.131.0)
Requirement already satisfied: Flask-Babel<4,>=3.1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (3.1.0)
Requirement already satisfied: psutil<6,>=5.9.8 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (5.9.8)
Requirement already satisfied: zipstream-ng<2.0.0,>=1.7.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (1.7.1)
Requirement already satisfied: feedparser<7,>=6.0.11 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (6.0.11)
Requirement already satisfied: Flask-Limiter<4,>=3.5.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (3.5.0)
Requirement already satisfied: frozendict<3,>=2.4.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (2.4.2)
Requirement already satisfied: pyserial<4,>=3.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (3.5)
Requirement already satisfied: future<1,>=0.18.3 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (0.18.3)
Requirement already satisfied: Click<9,>=8.1.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (8.1.7)
Requirement already satisfied: class-doc<0.3,>=0.2.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (0.2.6)
Requirement already satisfied: emoji<3,>=2.10.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (2.11.1)
Requirement already satisfied: colorlog<7,>=6.7.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (6.7.0)
Requirement already satisfied: pylru<2,>=1.2.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (1.2.1)
Requirement already satisfied: PyYAML<7,>=6.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (6.0.1)
Requirement already satisfied: sentry-sdk<2,>=1.40.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (1.45.0)
Requirement already satisfied: regex in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (2020.11.13)
Requirement already satisfied: watchdog<3,>=2.3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (2.3.1)
Requirement already satisfied: werkzeug<2.3,>=2.2.3 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (2.2.3)
Requirement already satisfied: unidecode in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (0.4.21)
Requirement already satisfied: semantic-version<3,>=2.10.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (2.10.0)
Requirement already satisfied: passlib<2,>=1.7.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==5.0.0) (1.7.4)
Requirement already satisfied: typing-extensions>=4.2.0 in /home/pi/oprint/lib/python3.7/site-packages (from pydantic==1.10.16->OctoPrint->OctoRelay==5.0.0) (4.2.0)
Requirement already satisfied: argon2-cffi-bindings in /home/pi/oprint/lib/python3.7/site-packages (from argon2-cffi>=23.1.0->OctoPrint->OctoRelay==5.0.0) (21.2.0)
Requirement already satisfied: more-itertools>=5.0.0 in /home/pi/oprint/lib/python3.7/site-packages (from class-doc<0.3,>=0.2.6->OctoPrint->OctoRelay==5.0.0) (9.1.0)
[... cut ...] n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-2o9bz6nq/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/oprint/include/site/python3.7/lgpio Check the logs for full command output.
Running setup.py install for lgpio: finished with status 'error'

The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.

RobinTail, 2024-07-11:
Python 3.7 ? @Orlando522 

[image]

RobinTail, 2024-07-11:
@Orlando522 , version 5 requires Python version at least 3.9.
I tried to express it in the release notes and the documentation:

[image]

Orlando522, 2024-07-11:
Hi,
It's on Raspberry Pi 3 Model B Rev 1.2
OctoPi OS version    : 0.18.0
OctoPrint version : 1.10.2
I install Plugin by notification in OctoPrint.

RobinTail, 2024-07-11:
The issue with `swig` can also be due to the OS.
0.18.0 is running `buster`, while 1.0.0 runs `bullseye`

RobinTail, 2024-07-11:
Could you upgrade your OS to Octopi 1.0.0, @Orlando522 ?
https://github.com/guysoft/OctoPi/releases/tag/1.0.0

It has python 3.9 and I tested that `swig` works on it.

RobinTail, 2024-07-11:
Another recommended solution, @Orlando522  && @ch3p4ll3 

```sh
apt-get remove swig
apt-get install swig3.0
ln -s /usr/bin/swig3.0 /usr/bin/swig
```

https://github.com/automl/auto-sklearn/issues/314#issuecomment-309682572

Orlando522, 2024-07-11:
OK, I understand ;-)

RobinTail, 2024-07-11:
@ch3p4ll3 , look, I found this:

https://pypi.org/project/swig/

Should we add this to the setup requirements?

RobinTail, 2024-07-11:
Yeah, I tried `pip install swig` and it works, adds the `swig` command, which is easier than `apt-get`

RobinTail, 2024-07-11:
@Orlando522 , I'm going to make a patch soon, that automatically installs `swig`.
If you will manage to upgrade your Python to 3.9, then it will be possible for you to install 5.0.0

RobinTail, 2024-07-11:
OK. I found out that when having python 3.9 it does not even need `swig`

```
pi@pi:~ $ pip install lgpio --force-reinstall --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting lgpio
  Downloading lgpio-0.2.2.0-cp39-cp39-linux_armv7l.whl (376 kB)
```

It downloads an exisitng wheel.

But with python 3.7 it's building that wheel, and fails because 3.7 is not supported

> ```
> Building wheel for lgpio (setup.py): started
> Building wheel for lgpio (setup.py): finished with status 'error'
> ```

Therefore, I'd say that no changes required beyond the already declared system requirements, @ch3p4ll3

ch3p4ll3, 2024-07-11:
> What do you think?

Yeah, I noticed the same on my Pi3 & Pi5. Btw, I agree, the system requirements are enough to avoid this error