borisbu/OctoRelay #287: Doesn't work after update to 5.0

iduarte1966, 2024-07-12:
After update to 5.0 I've lost the icons on dashboard, can't use this plugin anymore.
OctoPrint 1.10.2 Python 3.9.2 OctoPi* 1.0.0 (build 2023.10.09.151442

[image]
[image]

iduarte1966, 2024-07-12:
Solved.
Had to clear plugin data and setting each relay again.

[image]

closed 2024-07-12 by iduarte1966

RobinTail, 2024-07-12:
Interesting.
Is there a chance that you still have logs for that case, @iduarte1966 ?

RobinTail, 2024-07-12:
yes, @iduarte1966 
They are in this file:
[image]

iduarte1966, 2024-07-12:
> yes, @iduarte1966 <https://github.com/iduarte1966>
> They are in this file:
> image.png (view on web)

Seems that I don't have that "logging" menu
[image: image.png]
Ismael Duarte