borisbu/OctoRelay #299: Physical Button Plugin broken

ccle610, 2024-08-07:
Plugin Version
5.0.1

Question
So far I have used "Physical Button"a GPIO Input via Hardware Button to toggle the light ( =>Action: GPIO toggle).

The output is also used by the Octorelay plug-in to control the light. This worked well for a long time. Unfortunately, the button no longer reacts as long as the Octo Relay plug-in is running. Without the Octorelay plug-in, everything works perfectly!
I also tried the Workaround with GCODE in Physical Button, but also this does not work.

Is this a topic for here, or for Physicl Button?

Additional context
Physical Button Version 0.6.2

RobinTail, 2024-08-07:
yeah, I made a notice about it...

RobinTail, 2024-08-07:
Here is the solution for that problem, @ccle610 : 

https://github.com/borisbu/OctoRelay/issues/263#issuecomment-2241546096

ccle610, 2024-08-07:
Thx so much for this hint !

closed 2024-08-07 by ccle610

RobinTail, 2024-10-12:
There is probably a way to bypass the lock (exclusive pin reservation) established by `gpiozero` with a single line, that I'm currently investigating here, trying to make a cleanest solution possible:
https://github.com/borisbu/OctoRelay/pull/315

RobinTail, 2024-10-26:
Compatibility will be fixed in v5.1.0

PR #321 (2024-10-26): 🚀 5.1.0
Fixes #263 
Fixes #299 

- #320 
- #322 
- #315