borisbu/OctoRelay #304: Icons and relay controls do not appear when using Raspberry Pi 5.

corndog2000, 2024-08-18:
I have two relays turned on in OctoPrint settings. I do not see any buttons in OctoPrint to control these relays.

OctoPrint installed on Raspberry Pi 5 using this script: [https://github.com/paukstelis/octoprint_deploy](https://github.com/paukstelis/octoprint_deploy)

OctoPrint verison: 1.10.2
OctoRelay Version: 5.0.2

Operating system version:
[image]

Screenshots:
[image]
[image]
[image]

corndog2000, 2024-08-18:
Octoprint log file:
[octoprint-logs.zip](https://github.com/user-attachments/files/16646920/octoprint-logs.zip)

RobinTail, 2024-08-18:
There is an issue with your `lgpio` installation, @corndog2000 

```
2024-08-17 21:21:25,508 - py.warnings - WARNING - /home/pi/OctoPrint/lib/python3.11/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from lgpio: [Errno 2] No such file or directory: '.lgd-nfy-3'
```

Therefore `gpiozero` is trying an alternative approach with `rgpio` but it's failing too

```
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/gpiozero/pins/rpigpio.py", line 101, in __init__
    GPIO.setup(self._number, GPIO.IN, self.GPIO_PULL_UPS[self._pull])
RuntimeError: Cannot determine SOC peripheral base address
```

I'd suggest to make ensure `lgpio` python library to be installed properly.

```shell
pip3 install lgpio
```

That package should have been installed along with OctoRelay, but maybe there is an issue.
Try re-installing it manually, @corndog2000

RobinTail, 2024-08-18:
I also find it suspicious that the images are broken on the page.

[image]
 
Probably there is something wrong with the environment.
Consider a nightly build of [OctoPi](https://github.com/guysoft/OctoPi)

RobinTail, 2024-08-18:
CC @ch3p4ll3

ch3p4ll3, 2024-08-18:
Hi @corndog2000 what do u use to start octoprint? Systemd? In this case add `WorkingDirectory=/home/pi/OctoPrint` in the **Service** section of the octoprint unit file. Reload the services with `sudo systemct reload-daemons` and then try restarting octoprint with `sudo systemct restart octoprint`

corndog2000, 2024-08-18:
After trying your suggestions I still have errors in the OctoPrint log. 

I already had lpgio installed it seems.

```bash
pi@rpi5:~/.PrusaMK4 $ pip3 install lgpio --break-system-packages
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: lgpio in /usr/lib/python3/dist-packages (0.2.2.0)
```

I have also added the WorkingDirectory to the OctoPrint systemd entry.

```bash
pi@rpi5:~/.PrusaMK4 $ cat /etc/systemd/system/PrusaMK4.service
[Unit]
Description=The snappy web interface for your 3D printer
After=network.online.target
Wants=network.online.target

[Service]
Environment="PORT=5001"
Environment="BASEDIR=/home/pi/.PrusaMK4"
Environment="CONFIGFILE=/home/pi/.PrusaMK4/config.yaml"
User=pi
ExecStart=/home/pi/OctoPrint/bin/octoprint serve --config=${CONFIGFILE} --basedir=${BASEDIR} --port=${PORT}
WorkingDirectory=/home/pi/.PrusaMK4

[Install]
WantedBy=multi-user.target
pi@rpi5:~/.PrusaMK4 $
```

Here is the new log file:
[octoprint.log](https://github.com/user-attachments/files/16649286/octoprint.log)

ch3p4ll3, 2024-08-18:
Try changing GPIO, GPIO4 gave me problems too

RobinTail, 2024-08-18:
I'm concerned about these two messages:

1. `Defaulting to user installation because normal site-packages is not writeable`
2. `2024-08-17 21:21:25,508 - py.warnings - WARNING - /home/pi/OctoPrint/lib/python3.11/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from lgpio: [Errno 2] No such file or directory: '.lgd-nfy-3'`

I believe they are connected and related to some permission errors required for normal operation.

corndog2000, 2024-08-18:
> Try changing GPIO, GPIO4 gave me problems too

GPIO 4 is only used on the Lights option. It has been disabled since I've installed OctoRelay. My two enabled relays are pins 13 and 26.

[image]

corndog2000, 2024-08-18:
> I'm concerned about these two messages:
> 
>     1. `Defaulting to user installation because normal site-packages is not writeable`
> 
>     2. `2024-08-17 21:21:25,508 - py.warnings - WARNING - /home/pi/OctoPrint/lib/python3.11/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from lgpio: [Errno 2] No such file or directory: '.lgd-nfy-3'`
> 
> 
> I believe they are connected and related to some permission issues disrupting the normal operation.

That first message is due to this new message when using pip on Rasbian 12.

```bash
pi@rpi5:~/.PrusaMK4 $ pip3 install lgpio
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
pi@rpi5:~/.PrusaMK4 $
```

corndog2000, 2024-08-18:
python3-lgpio is also already installed.

```bash
pi@rpi5:~/.PrusaMK4 $ sudo apt install python3-lgpio
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-lgpio is already the newest version (0.2.2-1~rpt1).
The following package was automatically installed and is no longer required:
  libcamera0.2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@rpi5:~/.PrusaMK4 $
```

ch3p4ll3, 2024-08-18:
> GPIO 4 is only used on the Lights option. It has been disabled since I've installed OctoRelay. My two enabled relays are pins 13 and 26. [...]

Ok, but I see that the light is still active. Try disable it or change the pin

RobinTail, 2024-08-18:
Deactivate and restart OctoPrint, @corndog2000

corndog2000, 2024-08-18:
I activated it just to show where GPIO 4 is being used. Like I said, it has been disabled the entire time I've been experiencing issues.

ch3p4ll3, 2024-08-18:
Try stopping the service and starting it manually?

ch3p4ll3, 2024-08-18:
> ```shell
> ExecStart=/home/pi/OctoPrint/bin/octoprint serve --config=${CONFIGFILE} --basedir=${BASEDIR} --port=${PORT}
> WorkingDirectory=/home/pi/.PrusaMK4
> ```

And also try changing the working directory to `/home/pi/OctoPrint`

corndog2000, 2024-08-18:
I had to change the relays using GPIO 4, GPIO 7, and GPIO 8 to other GPIO numbers to get it working.

RobinTail, 2024-08-18:
Interesting.
So it does work, but on other channels.
@ch3p4ll3 , do you know what's the reason behind it? Is there an existing issue at `gpiozero` about that problem?

ch3p4ll3, 2024-08-18:
> Interesting. So it does work, but on other channels. @ch3p4ll3 , do you know what's the reason behind it? Is there an existing issue at `gpiozero` about that problem? Or is it the board issue?

[image]
maybe 'cause GPIOs 4/7/8 have other purposes beyond being an IO?

ch3p4ll3, 2024-08-18:
@corndog2000 Do you have 1-Wire enabled? https://forums.raspberrypi.com/viewtopic.php?t=370952

RobinTail, 2024-08-18:
> maybe 'cause GPIOs 4/7/8 have other purposes beyond being an IO?

Could be. And those purposes selected by what or how?
is it configurable in `raspi-config`?

corndog2000, 2024-08-18:
> > maybe 'cause GPIOs 4/7/8 have other purposes beyond being an IO?
> 
> Could be. And those purposes selected by what or how?
> is it configurable in `raspi-config`?
> 
> At least `1-Wire` is there
> [image]
> 

Pins 7 and 8 are used in the SPI bus which is also there in raspi-config.

RobinTail, 2024-08-19:
Ok, then I believe we figured this out.
@ch3p4ll3 , thank you for the support.

closed 2024-08-19 by RobinTail