borisbu/OctoRelay #305: Not working on Pi 5 because of hardware changes

Tryggvi44, 2024-08-19:
RPi.GPIO input/output library does not work on the Raspberry Pi model 5. This is because the RPi Model 5 now has a seperate chip caled RP1 for controlling the GPIO header.

RobinTail, 2024-08-19:
that's why we made OctoRelay version 5 that uses `lgpio` on RPI 5 instead, @Tryggvi44 
Check it out.

Tryggvi44, 2024-08-19:
I have OctoRelay 5.0.2 and it is not working for me.  OctoPrint: 1.10.2 

2024-08-19 10:36:59,267 - octoprint.server - INFO - Added new permission from plugin octorelay: PLUGIN_OCTORELAY_SWITCH (needs: "Need(method='role', value='plugin_octorelay_switch')")
|  OctoRelay (5.0.2) = /home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint_octorelay
2024-08-19 10:37:03,005 - octoprint.plugins.octorelay - INFO - Starting the plugin
2024-08-19 10:37:03,017 - octoprint.plugin - ERROR - Error while calling plugin octorelay
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint_octorelay/__init__.py", line 75, in on_after_startup
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint_octorelay/__init__.py", line 214, in handle_plugin_event
    self.toggle_relay(index, target) # UI update conducted by the polling thread
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint_octorelay/__init__.py", line 244, in toggle_relay
    relay = Driver.ensure(pin, inverted)
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint_octorelay/driver.py", line 49, in ensure
    relay = cls(pin, inverted, pin_factory)
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint_octorelay/driver.py", line 14, in __init__
2024-08-19 10:37:08,943 - octoprint.plugin - ERROR - Error while calling plugin octorelay
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint_octorelay/__init__.py", line 181, in on_event
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint_octorelay/__init__.py", line 316, in update_ui
    relay = Driver.ensure(
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint_octorelay/driver.py", line 49, in ensure
    relay = cls(pin, inverted, pin_factory)
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint_octorelay/driver.py", line 14, in __init__

ch3p4ll3, 2024-08-19:
Hi, could u please provide more logs?

RobinTail, 2024-08-19:
Yeah, the exact error is missing in the provided sample. Should be more lines after it.

RobinTail, 2024-08-19:
And just in case, please tell us which GPIO pins you are using for you relays, @Tryggvi44 .
We solved today an issue related to particular pins: https://github.com/borisbu/OctoRelay/issues/304#issuecomment-2295352086

Tryggvi44, 2024-08-19:
I am using pin 22, when I enable a relay I get this

2024-08-19 12:18:20,347 - octoprint.plugins.octorelay - INFO - Saving the settings: {'r1': {'active': True, 'relay_pin'                                                                                     : '22', 'rules': {'PRINTING_STARTED': {'state': None}, 'PRINTING_STOPPED': {'state': None}, 'STARTUP': {'state': None}}                                                                                     }}
2024-08-19 12:18:20,363 - octoprint.server.api.settings - ERROR - Could not save settings for plugin OctoRelay (5.0.2)
Traceback (most recent call last):
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/gpiozero/pins/pi.py", line 101, in pin
    pin = self.pins[n]
          ~~~~~~~~~^^^
KeyError: 22

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint/server/api/settings.py", line 1234, in _saveSett                                                                                     ings
    plugin.on_settings_save(data["plugins"][plugin_id])
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint_octorelay/__init__.py", line 56, in on_settings_                                                                                     save
    self.update_ui()
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint_octorelay/__init__.py", line 316, in update_ui
    relay = Driver.ensure(
            ^^^^^^^^^^^^^^
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint_octorelay/driver.py", line 49, in ensure
    relay = cls(pin, inverted, pin_factory)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint_octorelay/driver.py", line 14, in __init__
    self.handle = LED(pin, pin_factory=pin_factory, initial_value=inverted)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/gpiozero/devices.py", line 108, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/gpiozero/output_devices.py", line 200, in __init__
    super(DigitalOutputDevice, self).__init__(
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/gpiozero/output_devices.py", line 83, in __init__
    super(OutputDevice, self).__init__(pin, pin_factory=pin_factory)
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/gpiozero/mixins.py", line 85, in __init__
    super(SourceMixin, self).__init__(*args, **kwargs)
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/gpiozero/devices.py", line 549, in __init__
    pin = self.pin_factory.pin(pin)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/gpiozero/pins/pi.py", line 103, in pin
    pin = self.pin_class(self, n)
          ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/gpiozero/pins/rpigpio.py", line 111, in __init__
    GPIO.setup(self.number, GPIO.IN, self.GPIO_PULL_UPS[self._pull])
RuntimeError: Cannot determine SOC peripheral base address

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>This is the error when the plugin is started on boot

2024-08-19 12:15:12,018 - octoprint.plugins.octorelay - INFO - Starting the plugin
2024-08-19 12:15:12,042 - octoprint.plugin - ERROR - Error while calling plugin octorelay
Traceback (most recent call last):
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/gpiozero/pins/pi.py", line 101, in pin
    pin = self.pins[n]
          ~~~~~~~~~^^^
KeyError: 4

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint/plugin/__init__.py", line 275, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint_octorelay/__init__.py", line 76, in on_after_startup
    self.update_ui()
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint_octorelay/__init__.py", line 316, in update_ui
[... 1 lines cut ...]
            ^^^^^^^^^^^^^^
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint_octorelay/driver.py", line 49, in ensure
    relay = cls(pin, inverted, pin_factory)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/octoprint_octorelay/driver.py", line 14, in __init__
    self.handle = LED(pin, pin_factory=pin_factory, initial_value=inverted)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/gpiozero/devices.py", line 108, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/gpiozero/devices.py", line 549, in __init__
    pin = self.pin_factory.pin(pin)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/gpiozero/pins/pi.py", line 103, in pin
    pin = self.pin_class(self, n)
          ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tryggvi/OctoPrint/lib/python3.11/site-packages/gpiozero/pins/rpigpio.py", line 111, in __init__
    GPIO.setup(self.number, GPIO.IN, self.GPIO_PULL_UPS[self._pull])
RuntimeError: Cannot determine SOC peripheral base address

ch3p4ll3, 2024-08-19:
IMHO your error is this: `2024-08-17 21:21:25,508 - py.warnings - WARNING - /home/pi/OctoPrint/lib/python3.11/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from lgpio: [Errno 2] No such file or directory: '.lgd-nfy-3'`.

If that's the problem this is the fix: Add `WorkingDirectory=<Your octoprint instance path>` in the `Service` section of the octoprint unit file. Reload the services with `sudo systemct reload-daemons` and then try restarting octoprint with `sudo systemct restart octoprint`

source: https://github.com/gpiozero/gpiozero/discussions/1153

Tryggvi44, 2024-08-19:
I had messed with this a lot and your fix was not working so I did a clean install and the error came up but your fixed worked.

Addes WorkingDirectory to /etc/systemd/system/octoprint.service
[Service]
Environment="PORT=5000"
Environment="BASEDIR=/home/tryggvi/.octoprint"
Environment="CONFIGFILE=/home/tryggvi/.octoprint/config.yaml"
User=tryggvi
**WorkingDirectory=/home/tryggvi/.octoprint**
ExecStart=/home/tryggvi/OctoPrint/bin/octoprint serve --config=${CONFIGFILE} --basedir=${BASEDIR} --port=${PORT}

then
sudo systemctl daemon-reload
sudo systemctl restart octoprint

Thank you.

closed 2024-08-19 by Tryggvi44