borisbu/OctoRelay #318: Cancel switch off pop-up not showing

ch3p4ll3, 2024-10-17:
if I remember correctly, a pop-up once came up so that the countdown could be stopped if the relay is set to turn off at the end of a print job. 

The pop-up is not displayed and this error comes up in the console: 

```
packed_core.js?59b6a2d1:1879 Error calling onDataUpdaterPluginMessage on view model OctoRelayViewModel : Incorrect locale information provided
RangeError: Incorrect locale information provided
    at new NumberFormat (<anonymous>)
    at formatDeadline (http://10.188.26.141:5000/static/webassets/packed_plugins.js?d298bfc0:31932:37)
    at showHints (http://10.188.26.141:5000/static/webassets/packed_plugins.js?d298bfc0:32049:30)
    at OctoRelayViewModel.onDataUpdaterPluginMessage (http://10.188.26.141:5000/static/webassets/packed_plugins.js?d298bfc0:32097:11)
    at callViewModelIf (http://10.188.26.141:5000/static/webassets/packed_core.js?59b6a2d1:1878:75)
    at http://10.188.26.141:5000/static/webassets/packed_core.js?59b6a2d1:1872:138
    at Pn (http://10.188.26.141:5000/static/webassets/packed_libs.js?409a9b49:30:530)
    at Function.<anonymous> (http://10.188.26.141:5000/static/webassets/packed_libs.js?409a9b49:51:66)
    at callViewModelsIf (http://10.188.26.141:5000/static/webassets/packed_core.js?59b6a2d1:1872:95)
    at callViewModels (http://10.188.26.141:5000/static/webassets/packed_core.js?59b6a2d1:1871:56)
```

RobinTail, 2024-10-18:
that's interesting

RobinTail, 2024-10-18:
1. could you check the value of `LOCALE` in browser console?

[image]

RobinTail, 2024-10-18:
`LOCALE` comes from backend into the initial template
https://github.com/OctoPrint/OctoPrint/blob/ad9ebcaa0ef91950652fb2561b7c06b79d95a455/src/octoprint/templates/initscript.jinja2#L30

ch3p4ll3, 2024-10-18:
> 1. could you check the value of `LOCALE` in browser console, @ch3p4ll3 ?
> 
> 
> ![image](https://private-user-images.githubusercontent.com/13189514/377724799-7ce634a1-a3f6-464d-bfcf-8bb21e15825b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjkyNDQwMTQsIm5iZiI6MTcyOTI0MzcxNCwicGF0aCI6Ii8xMzE4OTUxNC8zNzc3MjQ3OTktN2NlNjM0YTEtYTNmNi00NjRkLWJmY2YtOGJiMjFlMTU4MjViLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDE4VDA5MjgzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlmMmM2MDZlMjBkNzZhNWQ5OTg5MDU2MDY0Y2JjYzhlMTI0NjUxZmQ0YTkzMDFiNDNiZTJlMTg5NGI0MWJhZDYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.R4d7K4I31Xmp9b87MZG2x__CnZHe4L-ke9GN1sclUYI)
> 
>     2. what is your browser? version?

1. The locale is set to `en`, but that's not the browser I use usually with octoprint
2. currently I'm using Firefox 128.3.1 on debian, usually I'm on Win11 with the latest chrome version

I don't know if it helps but in the pc I usually use(the one with W11) I have the system language in English but some regional settings(like time/date format, keyboard layout, etc...) in Italian

RobinTail, 2024-10-18:
> set to `en`

hmmm...
It's strange that it does not work, @ch3p4ll3 

Could you please run the following code the console of that browser:

```js
new Intl.NumberFormat('en', {
    style: "unit",
    unitDisplay: "long",
    minimumFractionDigits: 1,
    maximumFractionDigits: 1,
    unit: "minute"
  }).format(5)
```

ch3p4ll3, 2024-10-18:
> > The locale is set to `en`
> 
> hmmm... It's strange that it does not work, @ch3p4ll3
> 
> if you sure that `LOCALE` (the **particular** global variable when you open OctoPrint) is set to `en`, then could you please run the following code the console of that browser:
> 
> ```js
> new Intl.NumberFormat('en', {
>     style: "unit",
>     unitDisplay: "long",
>     minimumFractionDigits: 1,
>     maximumFractionDigits: 1,
>     unit: "minute"
>   }).format(5)
> ```
> 
> the output should be `5.0 minutes`
> 
> and if it fails, also try this one:
> 
> ```js
> new Intl.NumberFormat(undefined, {
>     style: "unit",
>     unitDisplay: "long",
>     minimumFractionDigits: 1,
>     maximumFractionDigits: 1,
>     unit: "minute"
>   }).format(5)
> ```

Yep, the output is correct in both cases, this evening I'll try on my main PC

RobinTail, 2024-10-18:
> the output is correct in both cases

I don't understand the cause for the error then.
Or maybe `LOCALE` is not `en`, but something else.

RobinTail, 2024-10-18:
The only way to achieve that error is having incorrect value of `LOCALE`

RobinTail, 2024-10-18:
But I'm going to apply my fix for that case and we'll see how it will work for you, @ch3p4ll3 
Please wait...

RobinTail, 2024-10-18:
@ch3p4ll3 , please try to install version `5.0.3rc1` from the `Prerelease` channel, or using this URL:
```
https://github.com/borisbu/OctoRelay/releases/download/5.0.3rc1/release.zip
```

Let me know if it fixes the issue.

ch3p4ll3, 2024-10-18:
> @ch3p4ll3 , please try to install version `5.0.3rc1` from the `Prerelease` channel, or using this URL:
> 
> ```
> https://github.com/borisbu/OctoRelay/releases/download/5.0.3rc1/release.zip
> ```
> 
> Let me know if it fixes the issue.

So, I checked the locale on my main PC and it's `it_IT`, trying this code resulted in an exception:
```js
new Intl.NumberFormat('it_IT', {
    style: "unit",
    unitDisplay: "long",
    minimumFractionDigits: 1,
    maximumFractionDigits: 1,
    unit: "minute"
  }).format(5)
```

Btw with this prerelease the problem is fixed. Ty

RobinTail, 2024-10-18:
are you using this Italian localization pack by chance, @ch3p4ll3 ?
https://community.octoprint.org/t/translating-octoprint/21144/13
or something else that sets the `LOCALE` to `it_IT` ?

ch3p4ll3, 2024-10-18:
> are you using this Italian localization pack by chance, @ch3p4ll3 ?

Nope

ch3p4ll3, 2024-10-18:
> or something else that sets the `LOCALE` to `it_IT` ?

[image]
My octoprint language is set to "Autodetect from browser", Chrome is in english soo, idk

ch3p4ll3, 2024-10-18:
[image]
anyway, if you look at AVAILABLE_LOCALES you'll find out that there are 2 italian locales

RobinTail, 2024-10-18:
OK, thank you. In 5.1 (after we fix this issue) I will add another intermediate fallback that would try replacing `_` with `-` to make it compatible.

PR #319 (2024-10-18): Fix countdown localization issue
Fixes #318
PR #307 (2024-10-18): 🚀 5.0.3
Fixes #318 

- #306 
- #308 
- #309 
- #310 
- #311 
- #316 
- #319
closed 2024-10-18 by RobinTail