borisbu/OctoRelay #323: Package 'octorelay' requires a different Python: 3.7.3 not in '<4,>=3.9'

davy932, 2024-10-26:
See install log below.  Looks like wont install as the Python version on OctoPrint is newer than expected

Updating, please wait.

+++++++++++++++++++++++++++++++
Now updating OctoRelay to 5.0.3
+++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install https://github.com/borisbu/OctoRelay/releases/download/5.0.3/release.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/borisbu/OctoRelay/releases/download/5.0.3/release.zip
Downloading https://github.com/borisbu/OctoRelay/releases/download/5.0.3/release.zip (51 kB)
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from OctoRelay==5.0.3) (1.10.2)
ERROR: Package 'octorelay' requires a different Python: 3.7.3 not in '<4,>=3.9'
Requirement already satisfied: RPi.GPIO in /home/pi/oprint/lib/python3.7/site-packages (from OctoRelay==5.0.3) (0.7.1)

The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.

RobinTail, 2024-10-26:
> Looks like wont install as the Python version on OctoPrint is newer than expected

Yes, it won't, @davy932 .
Version 5 of the plugin really requires in that range.
I put that notice in every [release note of v5](https://github.com/borisbu/OctoRelay/releases/tag/5.0.3) and in [Readme/Requiremenet](https://github.com/borisbu/OctoRelay/blob/master/README.md#requirements).

If you can not upgrade your Python to at least 3.9, please do not try upgrading the plugin to v5 and continue using v4.
The latest v4 you can install manually from the following URL:

```
https://github.com/borisbu/OctoRelay/releases/download/4.2.1/release.zip
```

closed 2024-10-26 by RobinTail

davy932, 2024-10-26:
Many thanks! I’ll follow the link and install manually Sent from my iPhone

Closed #323 as completed.