borisbu/OctoRelay #324: Icons not showing in Top Icons section

SCH00N3R, 2024-11-06:
Hello,

I am running Octoprint on a raspberry pi 5 8GB. I deployed it with octoprint_deploy.

After installing OctoRelay, it seems to have installed but I cannot see any of the icons in the top section of the octoprint page. I have tried disabling and re-enabling two relays. I have tried to change the logos to something other than default. Nothing so far has worked.

Any ideas? 

Thank you!

Neal

RobinTail, 2024-11-06:
Need to see your logs, @SCH00N3R .
There might be some error.

- Please proceed to the "Logging" section of your OctoPrint settings
- In the "Logging levels" dropdown find "octoprint.plugins.octorelay" and set "DEBUG" level
- Save and restart OctoPrint
- Then go to "Logging" again and in the "Logs" list download the "octoprint.log" entry
- You may shorten it to the current date/launch only and provide it please for study

SCH00N3R, 2024-11-06:
I am running python version 3.11.2

RobinTail, 2024-11-06:
> File "/home/schooner/OctoPrint/lib/python3.11/site-packages/gpiozero/pins/rpigpio.py", line 111, in init

It seems that it's using RPI.GPIO, which it should not on Pi 5.
We had such issue solved before, let me find it for you...

RobinTail, 2024-11-06:
@SCH00N3R , please proceed to this thread: https://github.com/borisbu/OctoRelay/issues/305#issuecomment-2296354460

There was a solution regarding `WorkingDirectory`.
Please try it and let me know if it worked for you or not.
CC @ch3p4ll3

SCH00N3R, 2024-11-06:
I am not having much luck. My service file now looks like this:

[Unit]
Description=The snappy web interface for your 3D printer
After=network-online.target
Wants=network-online.target

[Service]
Environment="PORT=5000"
Environment="BASEDIR=/home/schooner/.SCH00N3RPI"
Environment="CONFIGFILE=/home/schooner/.SCH00N3RPI/config.yaml"
User=schooner
WorkingDirectory=/home/schooner/.SCH00N3RPI
ExecStart=/home/schooner/OctoPrint/bin/octoprint serve --config=${CONFIGFILE} --basedir=${BASEDIR} --port=${PORT}

[Install]
WantedBy=multi-user.target

I can confirm that this setting is being taken up. It is included when I do systemctl edit SCH00N3RPI (that's the octoprint service name assigned by octoprint deploy).

I have checked for typos about 17 times.

RobinTail, 2024-11-06:
Here are other suggestions we're familiar so far, @SCH00N3R :

1. Try changing `WorkingDirectory` to `/home/schooner/OctoPrint`
2. Try deactivating all relays except one and changing its GPIO pin
  a. It was reported that on Pi 5 some pins cause issues: https://github.com/borisbu/OctoRelay/issues/304#issuecomment-2295352086
3. Try disabling `1-Wire`: https://github.com/borisbu/OctoRelay/issues/304#issuecomment-2295367650

SCH00N3R, 2024-11-06:
Thank you for all of your help, @RobinTail . This still is not working for me. I tried your suggestions. I also purposely chose pins that did not have a second use when I wired up my relays. I use pins 5 and 6. I checked the pluginmanager log and it stated that lgpio was already installed when I installed OctoRelay. That may be because I reinstalled OctoRelay a few times.

ch3p4ll3, 2024-11-06:
Hi @SCH00N3R, I sugget you to try installing octoprint manually following [this](https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspberry-pi-os-debian/2337) guide. 

in my Pi5 I installed Raspbian bookworm 64bit Lite. I would say you can try the basic installation and try if Octorelay works by serving Octoprint with `~/OctoPrint/venv/bin/octoprint serve`

SCH00N3R, 2024-11-07:
The octoprint venv had an old version of gpiozero installed (gpiozero 1.6.2). To check this:

cd ~/OctoPrint 
python3 -m venv venv
pip list | grep gpio

Sample output:
gpiozero                         1.6.2
lgpio                            0.2.2.0

To upgrade gpiozero within the venv:

pip install --upgrade gpiozero

Once the upgrade was completed, I restarted octoprint and the relay icons now work!

I should also add for posterity that I am running on bookworm lite 64bit.

closed 2024-11-07 by SCH00N3R

RobinTail, 2024-11-07:
That's great that you solved it, @SCH00N3R .
And thank you for telling us how.

Perhaps we should require particular version of `gpiozero` in `setup.py` dependencies

SCH00N3R, 2024-11-07:
Hello @RobinTail

That may be a good idea. I'm not certain if this is consistent, but GPIO 5 was a no-go for me. The icons disappeared when I tried to use it for my printer. Switched to GPIO 22 and now it works as expected.