borisbu/OctoRelay #325: Error due to unused pins?

christophhagen, 2024-11-20:
I'm facing a consistent issue (already with past versions of OctoRelay), where my configuration leads to errors that prevent OctoRelay from starting correctly. 

## Environment

I've installed `OctoPi` an a Raspberry Pi 5 manually, using a virtual environment and pip.
The error was already present in version 4.x, and persists until the latest version 5.1.0.

## Observed errors

The following two errors appear in the log:

```
2024-11-20 14:59:01,671 - octoprint.plugin - ERROR - Error while calling plugin octorelay
Traceback (most recent call last):
  File "/ch/run/print/venv/lib/python3.12/site-packages/gpiozero/pins/pi.py", line 411, in pin
    pin = self.pins[info]
          ~~~~~~~~~^^^^^^
KeyError: PinInfo(number=24, name='GPIO8', names=frozenset({'WPI10', 'BOARD24', '8', 8, 'GPIO8', 'J8:24', 'BCM8'}), pull='', row=12, col=2, interfaces=frozenset({'gpio', '', 'i2c', 'uart', 'spi', 'dpi'}))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/ch/run/print/venv/lib/python3.12/site-packages/octoprint/plugin/__init__.py", line 275, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/ch/run/print/venv/lib/python3.12/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^
  File "/ch/run/print/venv/lib/python3.12/site-packages/octoprint_octorelay/__init__.py", line 181, in on_event
    self.update_ui()
  File "/ch/run/print/venv/lib/python3.12/site-packages/octoprint_octorelay/__init__.py", line 318, in update_ui
    relay = Driver.ensure(
            ^^^^^^^^^^^^^^
  File "/ch/run/print/venv/lib/python3.12/site-packages/octoprint_octorelay/driver.py", line 51, in ensure
    relay = cls(pin, inverted, pin_factory)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/ch/run/print/venv/lib/python3.12/site-packages/octoprint_octorelay/driver.py", line 14, in __init__
    self.handle = LED(pin, pin_factory=pin_factory, initial_value=inverted)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/ch/run/print/venv/lib/python3.12/site-packages/gpiozero/devices.py", line 108, in __call__
    self = super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/ch/run/print/venv/lib/python3.12/site-packages/gpiozero/output_devices.py", line 192, in __init__
    super().__init__(pin, active_high=active_high,
  File "/ch/run/print/venv/lib/python3.12/site-packages/gpiozero/output_devices.py", line 74, in __init__
    super().__init__(pin, pin_factory=pin_factory)
  File "/ch/run/print/venv/lib/python3.12/site-packages/gpiozero/mixins.py", line 75, in __init__
    super().__init__(*args, **kwargs)
  File "/ch/run/print/venv/lib/python3.12/site-packages/gpiozero/devices.py", line 553, in __init__
    pin = self.pin_factory.pin(pin)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/ch/run/print/venv/lib/python3.12/site-packages/gpiozero/pins/pi.py", line 413, in pin
    pin = self.pin_class(self, info)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/ch/run/print/venv/lib/python3.12/site-packages/gpiozero/pins/lgpio.py", line 126, in __init__
    lgpio.gpio_claim_input(
  File "/ch/run/print/venv/lib/python3.12/site-packages/lgpio.py", line 755, in gpio_claim_input
    return _u2i(_lgpio._gpio_claim_input(handle&0xffff, lFlags, gpio))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/ch/run/print/venv/lib/python3.12/site-packages/lgpio.py", line 458, in _u2i
    raise error(error_text(v))
lgpio.error: 'GPIO busy'
```

and a bit later in the startup process:

```
2024-11-20 14:59:01,911 - octoprint.plugin - ERROR - Error while calling plugin octorelay
Traceback (most recent call last):
  File "/ch/run/print/venv/lib/python3.12/site-packages/gpiozero/pins/pi.py", line 411, in pin
    pin = self.pins[info]
          ~~~~~~~~~^^^^^^
KeyError: PinInfo(number=24, name='GPIO8', names=frozenset({'WPI10', 'BOARD24', '8', 8, 'GPIO8', 'J8:24', 'BCM8'}), pull='', row=12, col=2, interfaces=frozenset({'gpio', '', 'i2c', 'uart', 'spi', 'dpi'}))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/ch/run/print/venv/lib/python3.12/site-packages/octoprint/plugin/__init__.py", line 275, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/ch/run/print/venv/lib/python3.12/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^
  File "/ch/run/print/venv/lib/python3.12/site-packages/octoprint_octorelay/__init__.py", line 76, in on_after_startup
    self.update_ui()
  File "/ch/run/print/venv/lib/python3.12/site-packages/octoprint_octorelay/__init__.py", line 318, in update_ui
    relay = Driver.ensure(
            ^^^^^^^^^^^^^^
  File "/ch/run/print/venv/lib/python3.12/site-packages/octoprint_octorelay/driver.py", line 51, in ensure
    relay = cls(pin, inverted, pin_factory)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/ch/run/print/venv/lib/python3.12/site-packages/octoprint_octorelay/driver.py", line 14, in __init__
    self.handle = LED(pin, pin_factory=pin_factory, initial_value=inverted)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/ch/run/print/venv/lib/python3.12/site-packages/gpiozero/devices.py", line 108, in __call__
    self = super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/ch/run/print/venv/lib/python3.12/site-packages/gpiozero/output_devices.py", line 192, in __init__
    super().__init__(pin, active_high=active_high,
  File "/ch/run/print/venv/lib/python3.12/site-packages/gpiozero/output_devices.py", line 74, in __init__
    super().__init__(pin, pin_factory=pin_factory)
  File "/ch/run/print/venv/lib/python3.12/site-packages/gpiozero/mixins.py", line 75, in __init__
    super().__init__(*args, **kwargs)
  File "/ch/run/print/venv/lib/python3.12/site-packages/gpiozero/devices.py", line 553, in __init__
    pin = self.pin_factory.pin(pin)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/ch/run/print/venv/lib/python3.12/site-packages/gpiozero/pins/pi.py", line 413, in pin
    pin = self.pin_class(self, info)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/ch/run/print/venv/lib/python3.12/site-packages/gpiozero/pins/lgpio.py", line 126, in __init__
    lgpio.gpio_claim_input(
  File "/ch/run/print/venv/lib/python3.12/site-packages/lgpio.py", line 755, in gpio_claim_input
    return _u2i(_lgpio._gpio_claim_input(handle&0xffff, lFlags, gpio))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/ch/run/print/venv/lib/python3.12/site-packages/lgpio.py", line 458, in _u2i
    raise error(error_text(v))
lgpio.error: 'GPIO busy'
```

## Error investigation

It's unclear why there is always an error with pin 24, even if my configuration doesn't contain this pin.
I've tried to change my configuration in several ways, and even the default configuration produces this error for me.

## Possible fix

I've looked at the source code, and the error disappears when I change the following code in the file [`octoprint_octorelay/__init__.py`](https://github.com/borisbu/OctoRelay/blob/9ddbdf6fbbabd2cf83e4b393a89f071fadc03286/octoprint_octorelay/__init__.py#L316), lines 314-316:

```python
for index in RELAY_INDEXES:
    active = bool(settings[index]["active"])
    relay = Driver.ensure(
```

to:

```python
for index in RELAY_INDEXES:
    active = bool(settings[index]["active"])
    if not active:
        continue
    relay = Driver.ensure(
```

I'm not entirely sure why this works, but it seems to prevent it from initialising pins that it should not touch (since they are inactive). With this fix my two relays work as expected.

Please let me know if you need additional details.

RobinTail, 2024-11-20:
Thank you for reporting this in a such informative way, @christophhagen 
I will take a look on it soon

RobinTail, 2024-11-20:
Seems to be the same as the issue https://github.com/borisbu/OctoRelay/issues/175
But it was either not properly addressed in https://github.com/borisbu/OctoRelay/pull/176
or caused by the changes of the new driver.

RobinTail, 2024-11-25:
Can not confirm that error in v4.
Moreover, in v4 the driver did actually nothing when instantiating `Relay()`.
The v4 workflow is different.
So, even if there was an error in v4, that might be another error, that should be reported and investiaged separately.
The logs above are only relevant to v5, which I'm going to fix.
CC @christophhagen

closed 2024-11-25 by RobinTail

RobinTail, 2024-11-25:
@christophhagen ,
should be fixed in v5.1.1

Thank you for the contribution.

christophhagen, 2024-11-26:
I can confirm that this fixed my issue, thank you very much for the quick resolution!

PR #327 (2024-11-25): 🚀 5.1.1
Fixes #325 

- #326