borisbu/OctoRelay #332: Update failed

bobs409, 2024-12-14:
I've tried a few times to update Octorelay but it keeps giving the update failed warning.  I managed to get to the log files and I believe it's not working because the Python3 version on my unit is 3.7.3

This is the error: ERROR: Package 'octorelay' requires a different Python: 3.7.3 not in '<4,>=3.9'

I tried and tried but cannot get this python updated. I even tried to install version 3.9.7 but it failed.

Could it be because octoprint uses raspbian buster os?  Can the newer version be installed on buster?

Thanks,

Bob

RobinTail, 2024-12-15:
Hello @bobs409 ,

Version 5 of this plugin requires Python 3.9 or higher. It's feature is the support of Raspberry Pi 5 using another driver that is having higher Python requirements.

If you're willing for some reason to upgrade the plugin to version 5, you'd need to upgrade your Python first, which is what the error is telling you.

> I tried and tried but cannot get this python updated. I even tried to install version 3.9.7 but it failed.

I can not imagine what exactly you tried and how exactly that failed.
Consider this guide:
https://github.com/cp2004/OctoPrint-Upgrade-To-Py3

> octoprint uses raspbian buster os? 

I think you mean [OctoPi](https://github.com/guysoft/OctoPi)
It's current version certainly has modern Python.

bobs409, 2024-12-15:
Thank you for the reply.

closed 2024-12-15 by bobs409