borisbu/OctoRelay #334: rpi 5

Timon05, 2025-01-06:
it somehow uses RPI.GPIO which is not sutable for RPI 5, idk what the problem is but its not working

RobinTail, 2025-01-07:
@Timon05 

- Please proceed to the "Logging" section of your OctoPrint settings
- In the "Logging levels" dropdown find "octoprint.plugins.octorelay" and set "DEBUG" level
- Save and restart OctoPrint
- Then go to "Logging" again and in the "Logs" list download the "octoprint.log" entry
- You may shorten it to the current date/launch only and provide it please for study

Timon05, 2025-01-07:
herese my logg Anna Bocharova ***@***.***> schrieb am 7. Jan. 2025 um 9:30: @Timon05 Please proceed to the "Logging" section of your OctoPrint settings In the "Logging levels" dropdown find "octoprint.plugins.octorelay" and set "DEBUG" level Save and restart OctoPrint Then go to "Logging" again and in the "Logs" list download the "octoprint.log" entry You may shorten it to the current date/launch only and provide it please for study
2025-01-07 18:13:33,655 - octoprint.startup - INFO - ******************************************************************************
2025-01-07 18:13:33,656 - octoprint.startup - INFO - Starting OctoPrint 1.10.3
2025-01-07 18:13:33,657 - octoprint.startup - INFO - ******************************************************************************
2025-01-07 18:13:33,738 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2025-01-07 18:13:33,745 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2025-01-07 18:13:33,745 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 1.1.1.1:53 is working
2025-01-07 18:13:33,745 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2025-01-07 18:13:33,748 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/OctoPrint/venv/lib/python3.11/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2025-01-07 18:13:34,532 - octoprint.plugin.core - INFO - Plugin OctoEverywhere! (3.6.5) is disabled.
2025-01-07 18:13:34,578 - octoprint.plugin.core - INFO - Plugin HomeAssistant Discovery (3.7.0) is disabled.
2025-01-07 18:13:34,581 - octoprint.plugin.core - INFO - Plugin OctoVox Plugin (0.1.6) is disabled.
2025-01-07 18:13:34,593 - octoprint.plugin.core - INFO - Plugin PiNozCam (1.0.3) is disabled.
2025-01-07 18:13:35,037 - octoprint.plugins.errortracking - INFO - Initialized error tracking
2025-01-07 18:13:35,056 - octoprint.plugin.core - INFO - Found 36 plugin(s) providing 32 mixin implementations, 78 hook handlers
2025-01-07 18:13:35,058 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2025-01-07 18:13:35,084 - octoprint.server - INFO - Intermediary server started
2025-01-07 18:13:35,086 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2025-01-07 18:13:35,091 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2025-01-07 18:13:35,092 - octoprint.server - INFO - Added new permission from plugin achievements: PLUGIN_ACHIEVEMENTS_VIEW (needs: "Need(method='role', value='plugin_achievements_view')")
2025-01-07 18:13:35,093 - octoprint.server - INFO - Added new permission from plugin achievements: PLUGIN_ACHIEVEMENTS_RESET (needs: "Need(method='role', value='plugin_achievements_reset')")
2025-01-07 18:13:35,093 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_SHOW (needs: "Need(method='role', value='plugin_action_command_notification_show')")
2025-01-07 18:13:35,093 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_CLEAR (needs: "Need(method='role', value='plugin_action_command_notification_clear')")
2025-01-07 18:13:35,093 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: "Need(method='role', value='plugin_action_command_prompt_interact')")
2025-01-07 18:13:35,093 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: "Need(method='role', value='plugin_announcements_read')")
2025-01-07 18:13:35,093 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: "Need(method='role', value='plugin_announcements_read'), Need(method='role', value='plugin_announcements_manage')")
2025-01-07 18:13:35,093 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: "Need(method='role', value='plugin_appkeys_admin')")
2025-01-07 18:13:35,093 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_GRANT (needs: "Need(method='role', value='plugin_appkeys_user')")
2025-01-07 18:13:35,093 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_CREATE (needs: "Need(method='role', value='plugin_backup_create')")
2025-01-07 18:13:35,093 - octoprint.server - INFO - Added new permission from plugin file_check: PLUGIN_FILE_CHECK_RUN (needs: "Need(method='role', value='plugin_file_check_run')")
2025-01-07 18:13:35,094 - octoprint.server - INFO - Added new permission from plugin firmware_check: PLUGIN_FIRMWARE_CHECK_DISPLAY (needs: "Need(method='role', value='plugin_firmware_check_display')")
2025-01-07 18:13:35,094 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: "Need(method='role', value='plugin_logging_manage')")
2025-01-07 18:13:35,094 - octoprint.server - INFO - Added new permission from plugin octorelay: PLUGIN_OCTORELAY_SWITCH (needs: "Need(method='role', value='plugin_octorelay_switch')")
[... cut ...] ameter_set': {'type': 'unknown'}, 'slice_param_array': {'type': 'unknown'}, 'hevc_scaling_matrix': {'type': 'unknown'}, 'hevc_decode_parameters': {'type': 'unknown'}, 'hevc_decode_mode': {'type': 'menu', 'values': [{'value': '0', 'desc': 'Slice-Based'}], 'min': '0', 'max': '0', 'default': '0', 'value': '0'}, 'hevc_start_code': {'type': 'menu', 'values': [{'value': '0', 'desc': 'No Start Code'}, {'value': '1', 'desc': 'Annex B Start Code'}], 'min': '0', 'max': '1', 'default': '0', 'value': '0'}}}

RobinTail, 2025-01-07:
OK. This is the error when it's trying to use `lgpio`:

```
packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from lgpio: [Errno 2] No such file or directory: '.lgd-nfy-3'
```

RobinTail, 2025-01-07:
@Timon05 , please refer to these findings:

- https://github.com/borisbu/OctoRelay/issues/305#issuecomment-2296361146
- and https://github.com/borisbu/OctoRelay/issues/304

There can be several reasons for that issue and at least one advice from those threads should help you.
Let me know if you'd be able to fix it.

Timon05, 2025-01-07:
the 305 fixed it thanks:) Anna Bocharova ***@***.***> schrieb am 7. Jan. 2025 um 19:54: @Timon05 , please refer to these findings: #305 (comment) and #304 There can be several reasons for that issue and at least one advice from those threads should help you. Let me know if you'd be able to fix it.

closed 2025-01-07 by RobinTail