borisbu/OctoRelay #346: Updating to v5.1.2: cannot find -llgpio

gunderbolt, 2025-05-02:
OctoRelay is failing to update to v5.1.2 due to the `lgpio` wheel failing to build.

Here is my setup:

OctoPi 1.0.0 64bit
Raspberry Pi 4B - 2 GiB Ram
OctoPrint 1.11.0
Python 3.9.2

Here is the full output of the `plugin_softwareupdate_console.txt` log when trying to do the update:

```
2025-04-30 23:23:26,342   /home/pi/oprint/bin/python -m pip --disable-pip-version-check install https://github.com/borisbu/OctoRelay/releases/download/5.1.2/release.zip
2025-04-30 23:23:28,596 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2025-04-30 23:23:28,597 > Collecting https://github.com/borisbu/OctoRelay/releases/download/5.1.2/release.zip
2025-04-30 23:23:28,597 > Downloading https://github.com/borisbu/OctoRelay/releases/download/5.1.2/release.zip (52 kB)
2025-04-30 23:23:28,597 > Preparing metadata (setup.py): started
2025-04-30 23:23:29,661 > Preparing metadata (setup.py): finished with status 'done'
2025-04-30 23:23:29,661 > Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.9/site-packages (from OctoRelay==5.1.2) (1.11.0)
2025-04-30 23:23:29,662 > Requirement already satisfied: RPi.GPIO in /home/pi/oprint/lib/python3.9/site-packages (from OctoRelay==5.1.2) (0.7.1)
2025-04-30 23:23:29,662 > Requirement already satisfied: gpiozero in /home/pi/oprint/lib/python3.9/site-packages (from OctoRelay==5.1.2) (2.0)
2025-04-30 23:23:30,664 > Collecting lgpio (from OctoRelay==5.1.2)
2025-04-30 23:23:30,665 > Using cached lgpio-0.2.2.0.tar.gz (90 kB)
2025-04-30 23:23:30,666 > Preparing metadata (setup.py): started
2025-04-30 23:23:32,193 > Preparing metadata (setup.py): finished with status 'done'
2025-04-30 23:23:32,194 > Requirement already satisfied: colorzero in /home/pi/oprint/lib/python3.9/site-packages (from gpiozero->OctoRelay==5.1.2) (2.0)
2025-04-30 23:23:32,194 > Requirement already satisfied: OctoPrint-FileCheck>=2024.11.12 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (2024.11.12)
2025-04-30 23:23:32,195 > Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (2021.10.11)
2025-04-30 23:23:32,195 > Requirement already satisfied: OctoPrint-PiSupport>=2023.10.10 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (2023.10.10)
2025-04-30 23:23:32,195 > Requirement already satisfied: argon2-cffi>=23.1.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (23.1.0)
2025-04-30 23:23:32,196 > Requirement already satisfied: Babel<2.17,>=2.16 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (2.16.0)
2025-04-30 23:23:32,196 > Requirement already satisfied: cachelib<0.14,>=0.13.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (0.13.0)
2025-04-30 23:23:32,196 > Requirement already satisfied: Click<9,>=8.1.8 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (8.1.8)
2025-04-30 23:23:32,196 > Requirement already satisfied: colorlog<7,>=6.9.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (6.9.0)
2025-04-30 23:23:32,196 > Requirement already satisfied: emoji<3,>=2.14.1 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (2.14.1)
2025-04-30 23:23:32,196 > Requirement already satisfied: feedparser<7,>=6.0.11 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (6.0.11)
2025-04-30 23:23:32,197 > Requirement already satisfied: filetype<2,>=1.2.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (1.2.0)
2025-04-30 23:23:32,197 > Requirement already satisfied: Flask-Assets<3,>=2.1.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (2.1.0)
2025-04-30 23:23:32,197 > Requirement already satisfied: Flask-Babel<5,>=4.0.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (4.0.0)
2025-04-30 23:23:32,197 > Requirement already satisfied: Flask-Login<0.7,>=0.6.3 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (0.6.3)
2025-04-30 23:23:32,197 > Requirement already satisfied: Flask-Limiter<3.9,>=3.8 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (3.8.0)
2025-04-30 23:23:32,197 > Requirement already satisfied: flask<3.1,>=3.0.3 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (3.0.3)
2025-04-30 23:23:32,197 > Requirement already satisfied: frozendict<3,>=2.4.6 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (2.4.6)
2025-04-30 23:23:32,197 > Requirement already satisfied: limits<5 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (4.2)
2025-04-30 23:23:32,198 > Requirement already satisfied: markdown<3.8,>=3.7 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (3.7)
2025-04-30 23:23:32,198 > Requirement already satisfied: netaddr<1.4,>=1.3.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (1.3.0)
2025-04-30 23:23:32,198 > Requirement already satisfied: netifaces<0.12,>=0.11.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (0.11.0)
2025-04-30 23:23:32,198 > Requirement already satisfied: packaging in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (21.3)
2025-04-30 23:23:32,198 > Requirement already satisfied: libpass<2,>=1.8.1 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (1.9.0)
2025-04-30 23:23:32,198 > Requirement already satisfied: pathvalidate<4,>=3.2.1 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (3.2.3)
2025-04-30 23:23:32,198 > Requirement already satisfied: psutil<7,>=6.1.1 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (6.1.1)
2025-04-30 23:23:32,198 > Requirement already satisfied: pydantic<3,>=2.10.6 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (2.11.3)
2025-04-30 23:23:32,199 > Requirement already satisfied: pylru<2,>=1.2.1 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (1.2.1)
2025-04-30 23:23:32,199 > Requirement already satisfied: pyserial<4,>=3.5 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (3.5)
2025-04-30 23:23:32,199 > Requirement already satisfied: pytz in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (2023.3)
2025-04-30 23:23:32,199 > Requirement already satisfied: PyYAML<7,>=6.0.2 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (6.0.2)
2025-04-30 23:23:32,199 > Requirement already satisfied: requests<3,>=2.32.3 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (2.32.3)
2025-04-30 23:23:32,200 > Requirement already satisfied: sarge==0.1.7.post1 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (0.1.7.post1)
2025-04-30 23:23:32,200 > Requirement already satisfied: semantic_version<3,>=2.10.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (2.10.0)
2025-04-30 23:23:32,200 > Requirement already satisfied: sentry-sdk<3,>=2.20.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (2.27.0)
2025-04-30 23:23:32,201 > Requirement already satisfied: setuptools in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (44.1.1)
2025-04-30 23:23:32,201 > Requirement already satisfied: tornado<6.5,>=6.4.2 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (6.4.2)
2025-04-30 23:23:32,201 > Requirement already satisfied: watchdog<5,>=4.0.2 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (4.0.2)
2025-04-30 23:23:32,201 > Requirement already satisfied: websocket-client<1.9,>=1.8.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (1.8.0)
2025-04-30 23:23:32,202 > Requirement already satisfied: werkzeug<3.1,>=3.0.6 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==5.1.2) (3.0.6)
[... cut ...]
pi@nickpi:~ $ which swig
/usr/bin/swig
pi@nickpi:~ $ ls -l /usr/bin/swig
lrwxrwxrwx 1 root root 7 Nov 14  2020 /usr/bin/swig -> swig4.0
```

If I understand correctly, the `lgpio` library is built using the swig command. But I don't know why it's unable to find it.

I'm still using OctoRelay v4.2.0, which is working fine. But I would like to easily update all my OctoPrint plugins at once; this one is causing the whole thing to fail and forcing me to update the others one-at-a-time.

Thanks for taking a look!

RobinTail, 2025-05-02:
You can try this: https://github.com/borisbu/OctoRelay/issues/284#issuecomment-2222712950
Or try to install `lgpio` manually, so that the requirement would be satisfied, @gunderbolt 
Maybe using one of the prebuilt wheels: https://pypi.org/project/lgpio/#files

gunderbolt, 2025-05-03:
@RobinTail , thanks for the suggestions!

I had seen the comment that you linked, but my system already has `swig4.0`, and `swig3.0` is not in the package manager anymore. Also, a later [comment](https://github.com/automl/auto-sklearn/issues/314#issuecomment-1011921919) in the thread you linked to (from the 284 comment) suggested that v4.0 would also work (assuming the same root cause).

None of the prebuilt binaries for python 3.9 worked for me, unfortunately. They all failed immediately with a not-supported error, as seen in the following:

```
pi@nickpi:~ $ source oprint/bin/activate
(oprint) pi@nickpi:~ $ pip install lgpio-0.2.2.0-cp39-cp39-manylinux_2_34_aarch64.whl
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: lgpio-0.2.2.0-cp39-cp39-manylinux_2_34_aarch64.whl is not a supported wheel on this platform.
```

I tried to `sudo apt install libgpiod-dev`, but it didn't work and probably had nothing to do with the '-llgpio` option needed by the linker.

Looks like I may need to learn how swig works so I can figure out what dependency in the `lgpio` Python package is failing. Unfortunately, the `lgpio` package doesn't seem to have a repository where I can ask for help.

RobinTail, 2025-05-03:
Sorry to hear that. But you made a good research. I hope you'll find a solution.
In case you will, your findings could come helpful for #345 , where I'm getting the same error for Python 3.13 having no prebuilt wheel of `lgpio`.

gunderbolt, 2025-05-04:
I was able to get it working by installing the `liblgpio.so` C library directly. It may help you get a wheel for yourself for Python 3.13, but it will probably require a different approach to make it work for everybody.

The Python `lgpio` package's homepage points to <https://abyz.me.uk/lg/index.html> and the [Download](https://abyz.me.uk/lg/download.html) page has the installation instructions.

I already had the prerequisites (`swig`, `python3-dev` and `python3-setuptools`) installed, and proceeded with the other commands listed there:

```bash
wget http://abyz.me.uk/lg/lg.zip
unzip lg.zip
cd lg
make
sudo make install
```

I think this made the `-llgpio` linker flag work by building and installing `liblgpio.so`.

It also installed the `lgpio` Python library to the System. At this point, I considered just making a symlink from it to OctoPi's `~/oprint` virtualenv. But, I ended up activating the virtualenv and installing the package with pip:

```
source ~/oprint/bin/activate
pip install lgpio
```

For anyone interested, here is a zip of the wheel that it made: [lgpio-0.2.2.0-cp39-cp39-linux_aarch64.whl.zip](https://github.com/user-attachments/files/20026052/lgpio-0.2.2.0-cp39-cp39-linux_aarch64.whl.zip)

I was then able to update the OctoRelay plugin from within OctoPrint without any issues.

Downloading and installing the source from `lgpio`'s homepage in order to download and install the Python package from PyPI feels like a hack. If I keep looking into this, I would like to know
- what is inside the lgpio source file from PyPI,
- if it actually builds liblgpio.so, and
- if so, why it wasn't able to find it.

Thanks @RobinTail for your suggestions and help, and, most importantly, for all the work you've done on this really useful project!

closed 2025-05-04 by gunderbolt