borisbu/OctoRelay #347: error on boot and no icons in taskbar

Borreltje, 2025-05-03:
Raspberry Pi 4 Model B Rev 1.5 - 8gB memory
OctoPrint 1.11.0
Python 3.9.2

Hi All,
I am having trouble getting Octorelay working. I see these errors in the octoprint.log:

`
2025-05-03 13:10:58,614 - octoprint.plugin - ERROR - Error while calling plugin octorelay
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/gpiozero/pins/pi.py", line 411, in pin
    pin = self.pins[info]
KeyError: PinInfo(number=37, name='GPIO26', names=frozenset({'BOARD37', 'GPIO26', 'J8:37', 'WPI25', 26, 'BCM26', '26'}), pull='', row=19, col=1, interfaces=frozenset({'', 'spi', 'gpio', 'jtag', 'dpi', 'sdio'}))
`

I have another RPi running the samr octoprint version with the same octorelay version and that is running just fine.
I do not know what happened but there were no errors when I installed the plugin, but is is not showing the icons in the taskbar, I can configure the plugin in settings but nothing shows in the taskbar. I have configured three of the relays but nothing shows.

RobinTail, 2025-05-03:
could you provide more log surrounding that message, @Borreltje ?

RobinTail, 2025-05-03:
please check this out:
https://github.com/borisbu/OctoRelay/issues/236#issuecomment-1763334605
@Borreltje

Borreltje, 2025-05-03:
@RobinTail 

I will try a new card and do a new install. I will keep you updated on my findings. 

Thanks for now!

Borreltje, 2025-05-03:
@RobinTail 

Just did a complete re-install on a new SD. Everything works fine now. I did not think that a card would install at all when it is defective? But I am glad all is well now.

Thanks for helping me out!!

RobinTail, 2025-05-03:
I'm glad to read it, that you managed to solve the issue.
I'm not sure why it happens and manifests such a weird way, but probably `gpiozero` has some requirements to SD card, that older ones can not handle.

closed 2025-05-03 by RobinTail