borisbu/OctoRelay #353: OctoRelay on Raspberry PI5: No menubar icons

DJPerryCox, 2025-05-18:
I successfully installed OctoRelay on an Pi5 with Optopi 1.1.11 RC2. I can configure the relais but no buttons are displayed to interact. Octoprint.log indicates an issue with replacement of gpiozero by lgpio (see below). Any hints on how to solve this issue?

----

2025-05-18 19:48:29,985 - octoprint.plugins.octorelay - INFO - Saving the settings: {'r1': {'active': True, 'relay_pin': '25', 'rules': {'PRINTING_STARTED': {'state': None}, 'PRINTING_STOPPED': {'state': None}, 'STARTUP': {'state': None}}, 'show_upcoming': False}}
2025-05-18 19:48:29,999 - py.warnings - WARNING - /opt/octopi/oprint/lib/python3.11/site-packages/gpiozero/devices.py:300: PinFactoryFallback: **Falling back from lgpio: [Errno 2] No such file or directory: '.lgd-nfy-3'
  warnings.warn(**

2025-05-18 19:48:30,003 - octoprint.server.api.settings - ERROR - Could not save settings for plugin OctoRelay (5.2.0)
Traceback (most recent call last):
  File "/opt/octopi/oprint/lib/python3.11/site-packages/gpiozero/pins/pi.py", line 411, in pin
    pin = self.pins[info]
          ~~~~~~~~~^^^^^^
KeyError: PinInfo(number=22, name='GPIO25', names=frozenset({'GPIO25', 'J8:22', 'BCM25', 'BOARD22', 'WPI6', 25, '25'}), pull='', row=11, col=2, interfaces=frozenset({'', 'jtag', 'gpio', 'dpi', 'sdio', 'spi'}))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/octopi/oprint/lib/python3.11/site-packages/octoprint/server/api/settings.py", line 1283, in _saveSettings
    plugin.on_settings_save(data["plugins"][plugin_id])
  File "/opt/octopi/oprint/lib/python3.11/site-packages/octoprint/util/__init__.py", line 1683, in wrapper
    return f(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^
  File "/opt/octopi/oprint/lib/python3.11/site-packages/octoprint_octorelay/__init__.py", line 56, in on_settings_save
    self.update_ui()
  File "/opt/octopi/oprint/lib/python3.11/site-packages/octoprint_octorelay/__init__.py", line 318, in update_ui
    relay_state = Driver.ensure(pin, inverted).is_closed() if active else False
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/octopi/oprint/lib/python3.11/site-packages/octoprint_octorelay/driver.py", line 51, in ensure
    relay = cls(pin, inverted, pin_factory)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/octopi/oprint/lib/python3.11/site-packages/octoprint_octorelay/driver.py", line 14, in __init__
    self.handle = LED(pin, pin_factory=pin_factory, initial_value=inverted)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/octopi/oprint/lib/python3.11/site-packages/gpiozero/devices.py", line 108, in __call__
    self = super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/octopi/oprint/lib/python3.11/site-packages/gpiozero/output_devices.py", line 192, in __init__
    super().__init__(pin, active_high=active_high,
  File "/opt/octopi/oprint/lib/python3.11/site-packages/gpiozero/output_devices.py", line 74, in __init__
    super().__init__(pin, pin_factory=pin_factory)
  File "/opt/octopi/oprint/lib/python3.11/site-packages/gpiozero/mixins.py", line 75, in __init__
    super().__init__(*args, **kwargs)
  File "/opt/octopi/oprint/lib/python3.11/site-packages/gpiozero/devices.py", line 553, in __init__
    pin = self.pin_factory.pin(pin)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/octopi/oprint/lib/python3.11/site-packages/gpiozero/pins/pi.py", line 413, in pin
    pin = self.pin_class(self, info)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/octopi/oprint/lib/python3.11/site-packages/gpiozero/pins/rpigpio.py", line 101, in __init__
    GPIO.setup(self._number, GPIO.IN, self.GPIO_PULL_UPS[self._pull])
RuntimeError: Cannot determine SOC peripheral base address

RobinTail, 2025-05-18:
@DJPerryCox , we had 5 similar issues befoere:
https://github.com/search?q=repo%3Aborisbu%2FOctoRelay+Cannot+determine+SOC+peripheral+base+address&type=issues

Please check out the solutions

DJPerryCox, 2025-05-18:
I tried the WorkingDirectory change on the installed 64bit OS, without success. I then installed the suggested 32bit version, also without success, but when I added the WorkingDirectory on the 32bit version, it worked. Thx for the hints to possible solutions

DJPerryCox, 2025-05-20:
Actually, after several reboots an the installation of additional plugis, the OctoRelay Icons have vanished again. rejoiced too soon ;-((

DJPerryCox, 2025-06-07:
I now understand my problem, but I haven't solved it yet: I am running a script on the PI because I want to control the relays connected to the Pi (Waveshare 3 Relay HAT) with hardware buttons when necessary, for example to turn on the lights. This worked fine with GPIOZERO, as the pins are apparently not reserved there, which seems different with LGPIO on the PI5 (at least according to ChatGPT ;-). Whenever I activate my script, OctoRelay stops working and the icons disappear. In this respect, I can now use OctoRelay stably (currently the 32-bit version, I will check the 64-bit version again), but now I apparently no longer have the option of controlling the relays via a hardware button.

DJPerryCox, 2025-06-08:
@RobinTail Looks like that did the job. I modified the script to use the API and a first test worked without influencing the use of OctoRelay.

RobinTail, 2025-06-13:
this one seems to be solved

closed 2025-06-13 by RobinTail